scanse/sweep-3d-scanner #20: Use IMU to level the final scan

dcyoung, 2017-05-18:


drcpattison, 2017-08-17:
I have started investigating this issue by first trying to get the simpletest.py script to run using the following guide: https://learn.adafruit.com/bno055-absolute-orientation-sensor-with-raspberry-pi-and-beaglebone-black/overview

I have encountered two issues:
1. On this page: https://learn.adafruit.com/bno055-absolute-orientation-sensor-with-raspberry-pi-and-beaglebone-black/hardware it advises to disable the kernel's use of the serial port. Unfortunately I could not find option "A8 Serial" under advanced options within the raspi-config tool, it only went up to A6.

I discovered this is because the option is now contained within the Interfacing Options of the raspi-config main menu under P6.

2. After running simpletest.py I got the following error:
`serial.serialutil.SerialTimeoutException: Write timeout`

I discovered a quick fix to this solution is to add the following to `/boot/config.txt`

```
# Disable Bluetooth
dtoverlay=pi3-miniuart-bt # it disable the bluetooth interface
```

and then reboot- as detailed here: https://forums.adafruit.com/viewtopic.php?f=8&t=96471 so I guess that's the bluetooth functionality disabled then.

I'm imagining a solution now would be to pre-calibrate the sensor and hard code the results (so you don't have to wiggle the 3D scanner about while on site to calibrate the magnetometer), perhaps as detailed here: https://learn.adafruit.com/bno055-absolute-orientation-sensor-with-raspberry-pi-and-beaglebone-black/webgl-example#sensor-calibration (the results of which will vary for every user of course!) and then remove the while loop from simpletest.py  and either run the resulting script either at the start or the end of the 3D scan routine to obtain the pitch, roll and yaw.  The point cloud could then be orientated by applying rotation matrices to every point to align it correctly.

dcyoung, 2017-08-17:
Wonderful to see you're investigating this! I think others will be eager to dive in as well provided a little bit of starter content.

We could always create a guided calibration process with its own gui on the webapp. This calibration process could store results somewhere on the pi, so the scanner scripts could retrieve them later. I'm sure there are many solutions for this, but even a simple pickle file exported and imported back into python might suffice.  

This process would permit users to initialize or update the calibration whenever they like, before heading out to scan something.

dcyoung, 2017-09-19:
I did some testing this morning to get the IMU communicating data. Thanks @drcpattison for the advice mentioned above. I'll summarize the steps here:

## Raspi-Config Interface -> Serial settings: 
The provided sweep-3d-scanner image `v0.5.0` already includes the required raspi-config serial modifications. So if you are starting with a provided image, skip this section. However, if you are setting up a pi from scratch, or using your own image, you'll have to set the following:
From the command line, enter:
```bash
sudo raspi-config
```
Then, use the arrow keys in the menu to access Interfacing Options -> Serial, and:
- Disable the login shell access over serial (Interfacing Options -> Serial)
- Enable the serial port hardware (Interfacing Options -> Serial)

Then reboot

## Disable Bluetooth
By default, the main uart on the Pi is reserved for Bluetooth. While you can swap `AMA0` and `S0` aliases and force Bluetooth to use the inferior mini-uart, we'll just disable Bluetooth entirely to keep things simple. For additional info, check [here](https://spellfoundry.com/2016/05/29/configuring-gpio-serial-port-raspbian-jessie-including-pi-3/).

Completely disable bluetooth on the main uart so the BNO055 IMU can communicate over serial. Use sudo nano /boot/config.txt to add the following line to the bottom of `/boot/config.txt`:
```bash
# Disable Bluetooth so BNO055 IMU can use the serial interface
dtoverlay=pi3-disable-bt
```

## Install Adafruit Python BNO055 library
```bash
# prep
sudo apt-get update
sudo apt-get install build-essential python-pip python-dev python-smbus git

# install Adafruit python library for BNO055 IMU
cd /home/pi/code/scanse/
git clone https://github.com/adafruit/Adafruit_Python_BNO055.git
cd Adafruit_Python_BNO055
sudo python setup.py install
```

## Modify example for sweep-3d-scanner wiring
Open the simpletest.py script with:
```bash
nano /home/pi/code/scanse/Adafruit_Python_BNO055/examples/simpletest.py
```
Modify line \#34 to use the correct reset pin. Instead of GPIO pin 18, the sweep-3d-scanner uses GPIO pin \#4. Therefore, modify line \#34 to look like this:
```python
bno = BNO055.BNO055(serial_port='/dev/ttyAMA0', rst=4)
```

## Test script
Test the sensor using:
```bash
cd /home/pi/code/scanse/Adafruit_Python_BNO055/examples
sudo python simpletest.py
```

dcyoung, 2017-09-22:
As of release `v0.6.0` the provided Raspberry Pi image contains the modifications listed above. If you followed the wiring in the sweep-3d-scanner assembly guide (available [here](http://scanse.io/3d-scanning-kit/)), you should be able to jump right to testing the scripts:
```bash
cd /home/pi/code/scanse/Adafruit_Python_BNO055/examples
sudo python simpletest.py
```

ionutverzea, 2018-03-12:
Hi all,
I have some issues running the sensor. 
Before I start reading a new scanning, I save the position retrieved from the sensor, so far so good. If I made multiple scans, the position retrieved by the sensor is the same, even if I change the orientation of the device between scans... Have any of you experimenting this issue? Do you have a clue about how this could be fixed?

Example: 
1st scan: Heading 172, Roll 0, Pitch -170,0
2nd scan: Heading 175, Roll -2, Pitch 169  (device was rotated 90 degrees)
Even though for the second scan I would expect the heading to be either  270 or 90.

q3o, 2020-01-13:
Hi there,
We are finally setting up our 3D scanner and we got it to scan but I would totally love to be able to use the IMU to help on registering the scans.
Could anyone explain further which would be the whole process? would I need to shh the rasp, run simpletest then write those values down and go to the webapp and perform an scan? could there be a call on the webapp to perform the test and log it somewhere?
Thanks a lot and congratulations on this software and manuals, quite easy to follow even to one like me!