scanse/sweep-3d-scanner #28: Problems with running scan

drcpattison, 2017-05-29:
Hi there,

UK user here and proud owner of the Sweep. :)

I have assembled the 3d scanner (with firmware version 1.4 uploaded), loaded the pre-built software image to the Raspberry Pi (v0.2.2) and was able to successfully connect to the access point. I then ran all four tests which were successful. I have now started to take test scans using the default options (1hz motor, 500hz sample) which is where I have started to encounter some problems.

Once the scan has started there is a long period of time where the progress bar is stuck at 0% and then it will suddenly jump to a certain x% complete. At first I just thought there was a problem with it feeding the percentage back to the browser but with TestScan003 (attached) the progress bar ticked through every percentage from the late 80%s fully through to scan complete - with some random percentages appearing earlier on. This appears to coincide with the Scan Indexes recorded so there appears to be a deeper problem than simply browser feedback. TestScan001 appears to fail shortly after scanning starts.

I ran the above scans while outside the room because I didn't want to be in them.  Later on I did some more scans while inside the room to find out more info. When starting the scan the base rotates to find the limit switch, rotates a few degrees clockwise (looking down) and then a few degrees back to the limit switch - is this normal behaviour?  The first couple of scans I ran through my android smartphone on chrome and it seemed to hang on 0% with similar results to the TestScan001 file. I waited over 10 minutes one time and the stepper never moved although it was whining throughout and the percentage never increased.

For TestScan008 I ran it through my windows computer on chrome, again it was stuck on 0% for several minutes before jumping to random percentages however when the base moved it only incremented 1 step at a time, curiously the progress bar went over 100% and reached 156% at one stage with the progress bar and base incrementing intermittently. I waited about 5 mins after that but the scan never actually finished on the web app although the stepper motor did stop whining so it may have done.

I also got this error from the web app when trying to start the scan once:

```
main(argsdict)
  File "/home/pi/code/scanse/sweep-3d-scanner/scanner/scanner.py", line 258, in main
    with Sweep('/dev/ttyUSB0') as sweep:
  File "build/bdist.linux-armv7l/egg/sweeppy/__init__.py", line 114, in __enter__
RuntimeError: invalid response header checksum
```

Please let me know if there's any further information I could provide to help you diagnose the problem. I did not see any other error messages. I know how to SSH into the rpi if you need me to find anything in there.

Many thanks,
David

[image]
[Test Scan Files.zip](https://github.com/scanse/sweep-3d-scanner/files/1036518/Test.Scan.Files.zip)

dcyoung, 2017-05-29:
Hmmm. Definitely sounds like odd and unexpected behavior. Let's consider some possible causes, and try to gather some more information.

> I ran the above scans while outside the room because I didn't want to be in them

Until you get things working as expected, I'd recommend sticking around and watching the scanner's behavior. The base also rotates slow enough that you can always stand behind the 2D scan plane if you don't want to be in the scan.

#### Components
- Are you using the exact same parts from the parts list, or have you made any substitutions? For example, are you using a different stepper motor or different version of the Raspberry Pi?

#### Physical mounting
- Is the sweep sensor mounted in the 90 degree scanner bracket such that the wires coming out of the sweep face directly down? 

The photograph on page 14 of the current [assembly instructions](https://s3.amazonaws.com/scanse/3d_scanning_kit_assembly_guide.pdf) (`rev 0.93`) can be misleading, as it looks like the wires are pointing towards 3 o'clock. The sweep should be mounted such that the wires face downward (6 o'clock). This coincides with the LED/thin groove on the face of the sweep pointing directly up. When the sweep is powered on, the LED should be visible through the tiny hole at the top of the scanner bracket.

#### Sweep sensor
- Have you had any issues with the sweep sensor itself? 
- Does the sweep work just fine with the visualizer?
- Could you try steaming data in the visualizer while the sensor sits unperturbed. Judging by eye, are there hiccups in the data stream or any abnormal variance in the inter-arrival time between scans? If you look at the DataTable view, and scroll to the bottom of the index column, is the variability in point count between successive scans within say 10-20%? O does the point count occasionally double (ie: a scan with ~twice as many points as its temporal neighbors).

#### sweep-sdk
- Have you tried using `libsweep` or the `sweeppy` bindings before outside of the scanner project? If so have you experienced any odd behavior there?

#### expected + observed behavior
> When starting the scan the base rotates to find the limit switch, rotates a few degrees clockwise (looking down) and then a few degrees back to the limit switch - is this normal behaviour?

This is the expected behavior. The idea was for the motor to move quickly back to the home position (in case the base was ~180 degrees away) using larger less precise movements. Then, it would back off the limit switch slightly and move back onto the limit switch with smaller more precise movement to achieve a more reliable starting home position. This was the inteded behavior. Currently some issues with the motor control are still being sorted out, so the fidelity of the movement for each of those phases (long range quick return + short range slow return) are using the same type of movement. 

> Once the scan has started there is a long period of time where the progress bar is stuck at 0% and then it will suddenly jump to a certain x% complete

The expected behavior is as follows: The progress bar should be at 0% throughout the setup procedure (including reset, motor speed adjustment, and base reset) which can take ~30-60seconds. After this setup procedure, the webapp should indicate `Scan is in Progress` and the progress bar should then move while the scan is taking place.

- By "Once the scan has started" do you mean once the webapp indicates `Scan is in Progress`?
- Could you be more specific than "long period"? 
- During this time is the base rotating after each scan, or does it stay fixed at the home position?

> random percentages appearing earlier on. This appears to coincide with the Scan Indexes recorded so there appears to be a deeper problem than simply browser feedback.

You are correct in your interpretation. There is definitely a deeper issue here. The browser simply receives an indication of the current scan index and the expected number of total scans. It calculates a ratio from the two and uses it to update the progress bar. The browser has no concept of the correct sequence of things, it simply displays the information it is given. 

The deeper issue is that the `scanner.py` script seems to avoid outputting scans altogether. The fact that there are gaps in scan indices also hints at a bug, where `scanner.py` increments the scan index using the enumerated scan_count, rather than by incrementing a counter of validated scans which the script intends to output. Typically this never happens, or happens so rarely that it has been overlooked. But observing this behavior is exposing some bugs, which is good. I'll make a new separate issue for this. 

> the progress bar went over 100% and reached 156% at one stage with the progress bar and base incrementing intermittently

This one is definitely a bug. One guess is that the scanner is either missing the sync byte, and interpreting two rotations as a single large scan, or somehow returning more samples than should be possible in a single scan. When the `scanner.py` script accumulates scans, it checks for scans that are too large and simply discards them to try again before moving the base. https://github.com/scanse/sweep-3d-scanner/blob/44f9414df45210a7693225c3f8a6d69a113dda9b/scanner/scanner.py#L165
[... cut ...]
> Please let me know if there's any further information I could provide to help you diagnose the problem

In general, if an issue occurs that does not result in an error message propagted to the browser, you can SSH into the machine and retrieve the webapp's log file located at `/home/pi/scanner_output.log`. However, currently this has to be done before shutting down the pi or the log file will be overwritten. See #25

That's a lot to go through, but hopefully it gives you a few things to double check. Meanwhile, I'll try to get a fix for those other issues worked up ASAP.

dcyoung, 2017-05-29:
As mentioned previously, the `scanner.py` script will discard scans with sample counts above a certain threshold. This is meant to protect against accidental double scans due to missing a sync byte. 

However, occasionally a particularly lucky device will perform notably better than the advertised specifications (yield scans with higher than expected point counts). The effect would be exaggerated in an environment that produces high signal strength readings. See [this article](https://support.scanse.io/hc/en-us/articles/115006333327-Theory-of-Operation). I'm wondering if it is possible that your device is yielding such potent scans that the `scanner.py` script is misinterpreting them as 2 scans and then discarding them. 

To test this, can you:

- Open the visualizer
- Change the device motor speed/sample rate to match scan settings (1Hz/500Hz) 
- Start scanning
- Hold the device in the same orientation as when mounted in the scanner... such that it is scanning the same plane of the room.
- Look at the DataTable view in the Visualizer and scroll to the bottom to get an idea for the number of samples per scan on average
- Report back. 

It is unlikely this is the cause, but it would explain the behavior. Either way, the code should be changed such that scans are discarded on the basis of containing un-ordered samples (rollover), rather than too many samples.

drcpattison, 2017-05-30:
Hi David,

Thanks very much for your quick and detailed reply, it is very much appreciated. :)

>Components
>
>Are you using the exact same parts from the parts list, or have you made any substitutions? For example, >are you using a different stepper motor or different version of the Raspberry Pi?

I have tried to use the exact same parts where they can be easily source within the UK, for the stepper motor I can confirm I ordered this: "0.9deg Round Nema 14 Stepper Motor Bipolar 0.65A 12Ncm/17oz.in Φ36x19.5mm 4-wire" and I can confirm I am using the Raspberry Pi 3 Model B.
This shouldn't affect functionality but the parts below are the ones which I've needed to substitute because I couldn't find a supplier for them within the UK (I'm thinking this could be something for you guys to take in consideration if you ever wanted to modify the design):

1. 10000mAh Power Bank - not available from the Amazon UK site, also I could not find it anywhere else. Indeed, I found it impossible to find a square shaped power bank of this capacity (all seem to be smartphone shaped) so I'm using an external 12V lead acid battery with 12v to dual usb converter instead. I'm going to modify my design soon to include an energy meter so I can get a bit more info about the energy being consumed etc so not this isn't an issue for me but it might prove difficult for other Europeans who want to use your elegant self contained power supply solution.

2. 14mm Circular Bubble Spirit Level - again, I couldn't find a supplier who could provide this exact size. I have a workaround of putting a spirit level of a different size on top of the tripod first.  I was thinking perhaps the IMU could be used in future to either indicate somehow to the user about being on a level surface or even better it could compensate for a slightly unlevel surface and then transforming the scan by that error to make to make it level again?

3. SPDT AC Miniature Micro Switch - found a similar one off RS: "SPDT-NO/NC Simulated Roller Lever Microswitch, 5 A @ 125 V ac", works according to the tests so no issues there.

4. #'2-28 Thread Size, 3/4" Length Screws
5. #'2-28 Thread Size, 3/8" Length Couldn't find these types of rolling screws probably since they are in imperial dimensions, found something close enough in the end that seems to do the job.

The other thing that was my fault but serves as a warning to others is try not to buy the hex nut "1/4-20 Nut" with nylon in the middle as the nylon plastic melted around my soldering iron tip and ruined it!

> Physical mounting
>
>Is the sweep sensor mounted in the 90 degree scanner bracket such that the wires coming out of the >sweep face directly down?

I saw another user had an issue with this on the forums so I was careful to make sure it was pointing down during assembly. I don't want to take it apart again to fully confirm but I can see the blue LED status light of the Sweep is pointing upwards so hence the wire connector must be pointing down. :)

>Sweep sensor
>
>Have you had any issues with the sweep sensor itself?

No issues as far as I can tell, seems to be working just great.

>Does the sweep work just fine with the visualizer?

It seems to work very well within the visualiser so no complaints there.

>Could you try steaming data in the visualizer while the sensor sits unperturbed. Judging by eye, are >there hiccups in the data stream or any abnormal variance in the inter-arrival time between scans? If >you look at the DataTable view, and scroll to the bottom of the index column, is the variability in point >count between successive scans within say 10-20%? Or does the point count occasionally double (ie: a >scan with ~twice as many points as its temporal neighbors).

Within the point cloud I seem to getting be within 624-655 point counts per scan at 1hz motor, 500hz sample frequencies. I did not notice any double point counts after running for several minutes. No hiccups in the data stream or abnormal variance between inter-arrival times either.

>sweep-sdk
>
>Have you tried using libsweep or the sweeppy bindings before outside of the scanner project? If so >have you experienced any odd behavior there?

I haven't tried, for my purposes I'm just looking to take static 3D scans of terrain so I have no need to try working on anything upfront beyond getting the raw 3D scans. Most of my work will be in the post-processing of the scans.

>expected + observed behavior
>This is the expected behavior. ...

Ok that's good then!

>By "Once the scan has started" do you mean once the webapp indicates Scan is in Progress?

Correct. The setup procedure has almost always worked correctly - apart from that error one time I highlighted to you.

>Could you be more specific than "long period"?

You're talking about 3-8 mins when it would be stuck in the home position. I can do another test with more detailed timing observations if you need me to.

>During this time is the base rotating after each scan, or does it stay fixed at the home position?

It stays fixed in the home position. It only moved in conjunction to when the progress bar moved (sometimes to random percentages, sometimes incremented) when base did move it always stepped in increments during the scan though.

>In the future, if the process makes it to Scan is in Progress phase, you should expect base movement almost immediately. The base should move 1 step between each successive 2D scan. If it isn't moving, that means something is wrong.

Ok definitely something wrong then since it isn't moving almost immediately.

Thanks for the further info you've provided. Shall I update the recent changes you have made, if so do you have info on the best way to do this? If not I shall continue performing test scans and wait for the next pre-built image while checking `scanner_output.log` for any errors I can provide.

In relation to your second comment, I think I've answered that with 624-655 samples per scan as above, if not please let me know.

drcpattison, 2017-05-31:
Hi David,

Good news, after switching to v0.3.0 I'm pleased to report everything worked correctly first time, both in terms of base movement and on the webapp! Thanks so much for your help.

One small thing I've spotted is in the Visualiser you can see a small gap in the scan where it looks like it hasn't quite managed the full 360 degrees, this has happened in the first two scans I've made. Hopefully this just needs a small tweak in the code somewhere.
[TestScan010.zip](https://github.com/scanse/sweep-3d-scanner/files/1043082/TestScan010.zip)

Also in relation to this:

>I was thinking perhaps the IMU could be used in future to either indicate somehow to the user about being on a level surface or even better it could compensate for a slightly unlevel surface and then transforming the scan by that error to make to make it level again?

I see you've already raised this as an issue so please ignore. No promises but I might have look into this issue at some point as I've been doing a lot of work with a similar device recently so I could probably make a start on it at least.

dcyoung, 2017-06-01:
@drcpattison  I'm glad to hear things are working! The project will progress much more quickly with detailed bug reports like this. 

> TestScan010.zip

Definitely room for improvement in tightening up the scanner here, both in physical alignment and in software. But this is definitely making progress! Very neat to see people's point clouds, even at these early stages. It really conveys information about a space you can't get from pictures. By the way.... is there a door in that room??

> you can see a small gap in the scan

I've noticed this on our scans as well but haven't looked into debugging it just yet. I think it might be a simple issue with the calculation of how many scans to take.

>  I could probably make a start on it at least.

Looking forward to it! I think this would really help improve the process of aligning/merged multiple scans later.

closed 2017-06-01 by dcyoung