scanse/sweep-3d-scanner #29: Redirect to 3D scanner homepage before restart

drcpattison, 2017-05-29:
Hi there,

While performing 3d scans I have had to restart the rpi a few times. When you do this you obviously lose connection and have to reconnect to the wireless hotspot. Unfortunately when reconnecting the Chrome browser has a nasty habit (in this case) of refreshing the page upon reconnection so if the request_reboot page is still in the address bar the rpi will reboot again! I think this could be solved by redirecting the browser to the homepage before performing the reboot command.

Many thanks,
David

dcyoung, 2017-05-29:
Definitely agree. I've experienced that frustrating accidental shutdown as well. 

As you say, this could likely be a very easy fix. Did you want to take a stab at it, and submit a Pull Request? If not I'll try to get around to it soon.

drcpattison, 2017-05-30:
Good stuff, a simple one liner added to both functions fixes it for me :)
This is my first time using this site but hopefully I've submitted a pull request for you to review.

dcyoung, 2017-05-30:
Tested and works fine for me. For a first time user you've got it down! Pull request has been merged. Thanks.

closed 2017-05-30 by dcyoung

drcpattison, 2017-05-30:
So I've made a contribution to an open-source project - never thought I'd be able to say that!
You're welcome!