scanse/sweep-3d-scanner #35: Running from image works but after following update instructions, scanning just hangs

ewumnrr, 2017-07-17:
I followed instructions and executed the following sequence on a working scanner that had the most recent SD card image version:
===
pushd /home/pi/code/scanse/sweep-sdk
git pull origin master
sudo rm /usr/local/lib/libsweep*
cd libsweep
rm -Rf build
mkdir -p build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
cmake --build .
sudo cmake --build . --target install
sudo ldconfig
popd
sudo rm /home/pi/.local/lib/python2.7/site-packages/sweeppy*
pushd /home/pi/code/scanse/sweep-sdk/sweeppy
python setup.py install --user
popd
pushd /home/pi/code/scanse/sweep-3d-scanner
git pull origin master
npm install
popd
sudo reboot
====
No errors reported during this sequence.
After this update, the component test passes except for the scanner test. The test case just hangs.
Starting a scan via the web interface also just hangs at 0%.
Please verify that the published update sequence works.

dcyoung, 2017-07-17:
Looking into this a bit this morning, and it appears that `sweep-sdk` version `v1.2.1` introduced an issue on Raspberry Pi. Haven't yet determined what, but for now avoid updating `sweep-sdk`, or if you do keep it limited to `v1.2.0`. Additionally, there haven't been any major updates to sweep-3d-scanner either. So no need to update as of today.

dcyoung, 2017-07-18:
This has been fixed as of the latest `sweep-sdk` release: `v1.2.2`. I'm in the process of rebuilding a fresh image for the pi. Should be up in a few hours.

ewumnrr, 2017-07-19:
Just want to note that I went back and reflashed my SD card but that did not bring the scanning process back to work. Did the above procedure also reflash the LIDAR module and brick it?

dcyoung, 2017-07-19:
Could you be more specific ? Did you refresh the sd card completely with the previous from release v0.3.0 or v0.4.0? 

The sd card is simply a raspberry pi image, it does not affect the sweep in any way. Are you having trouble with the sweep device itself outside of trying to use it in your 3d scanner project ? Does the device work fine on a computer with the visualizer?

After reflashing what behavior are you experiencing ? The same hanging behavior as before? 
Can you connect to the pi with the webapp and run the other tests just fine ?

Did the sweep ever get unplugged ? Try unplugging it and plugging it back in after a reboot of the pi. Then ssh into the pi and run the libsweep examples (located in home/pi/code/scanse/sweep-sdk/libsweep/examples/build)  from the command line. Do those hang as well?

ewumnrr, 2017-07-19:
I had flashed back to 0.3.0. But when I started a scan, it was hanging, didn't start the actual scan.
As you suggested, I just connected the sweep device to my Mac and it is actually working fine.
After I connected it back to the pi, it is now working normally, so thanks for the suggestion.
No idea what fixed it. I had pulled both cables to the battery before, it was definitely completely powered down before.

ewumnrr, 2017-07-19:
Flashed SD card to 0.4 and it works as well. I then ran the update sequence as I cited above and it still works. Case closed. Thanks!

dcyoung, 2017-07-19:
Hmmm I would think a full power cycle would do it. But perhaps the usb port with the sweep has to be cycled in some way. Glad to hear it is working now!

closed 2017-07-19 by dcyoung